junkfix/esp32-ds18b20 #5: ESP-IDF 5.x

dima-sz, 2023-11-06:
Hi. Will there be a fix for version 3.0 (based on ESP-IDF 5.1) of the Arduino ESP32 core.

EvEggelen, 2023-11-17:
Can you indicate what you problem is ? Today I tested with With 6.4 ( https://github.com/platformio/platform-espressif32/releases/tag/v6.4.0 ) and it seems to work.

dima-sz, 2023-11-18:
I use arduino IDE 1.8.19 and
[image]

https://github.com/espressif/arduino-esp32

I get a compilation error

D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp: In constructor 'OneWire32::OneWire32(uint8_t, uint8_t, uint8_t, uint8_t)':
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:57:49: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
   57 |                                                 GPIO.enable_w1ts = (0x1 << owpin);
      |                                                 ^~~~
      |                                                 PI
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:59:49: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
   59 |                                                 GPIO.enable1_w1ts.data = (0x1 << (owpin - 32));
      |                                                 ^~~~
      |                                                 PI
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:63:58: error: 'GPIO_PIN_MUX_REG' was not declared in this scope; did you mean 'GPIO_PIN19_REG'?
   63 |                                         PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[owpin]);
      |                                                          ^~~~~~~~~~~~~~~~
      |                                                          GPIO_PIN19_REG
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:63:41: error: 'PIN_INPUT_ENABLE' was not declared in this scope
   63 |                                         PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[owpin]);
      |                                         ^~~~~~~~~~~~~~~~
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:64:41: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
   64 |                                         GPIO.pin[owpin].pad_driver = 1;
      |                                         ^~~~
      |                                         PI
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp: In member function 'bool OneWire32::reset()':
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:98:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
   98 |         GPIO.pin[owpin].pad_driver = 1;
      |         ^~~~
      |         PI
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp: In member function 'bool OneWire32::read(uint8_t&, uint8_t)':
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:142:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  142 |         GPIO.pin[owpin].pad_driver = 1;
      |         ^~~~
      |         PI
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp: In member function 'bool OneWire32::write(uint8_t, uint8_t)':
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:180:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  180 |         GPIO.pin[owpin].pad_driver = power_default? 0 : 1;
      |         ^~~~
      |         PI
exit status 1

htmltiger, 2023-11-19:
this may be linked with https://github.com/espressif/arduino-esp32/issues/8846
Try add this in OneWireESP32.h

```
#include "soc/gpio_struct.h"
```

dima-sz, 2023-11-20:
now another error

D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp: In constructor 'OneWire32::OneWire32(uint8_t, uint8_t, uint8_t, uint8_t)':
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:63:58: error: 'GPIO_PIN_MUX_REG' was not declared in this scope; did you mean 'GPIO_PIN19_REG'?
   63 |                                         PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[owpin]);
      |                                                          ^~~~~~~~~~~~~~~~
      |                                                          GPIO_PIN19_REG
D:\Arduino\ArduinoID\libraries\esp32-ds18b20\OneWireESP32.cpp:63:41: error: 'PIN_INPUT_ENABLE' was not declared in this scope
   63 |                                         PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[owpin]);
      |                                         ^~~~~~~~~~~~~~~~
exit status 1

htmltiger, 2023-11-21:
Please try
```
#include "soc/gpio_periph.h"
```

dima-sz, 2023-11-22:
Now everything works, thank you very much.

closed 2023-12-03 by htmltiger