junkfix/esp32-ds18b20 #7: Long distance (not so long) issue?

skillsfactory, 2024-01-12:
Hello,

I've tested your library on a breadboard with 4 sensors and 3 BUS because I've different distances.
In my real environment I can find the two closest sensors but not the two farthest (about 20 m on a single BUS).
I used an alarm BUS cable for the connections and with the original library I can reach them without particular problems, problems that I find instead on the two closest sensors, so I decided to try this other library...
Do you know the maximum distance that can be reached with this library?
Thanks
Luca

htmltiger, 2024-01-12:
you may need stronger pullup, 4.7k is standard so if you try 2.2k or less to check if it works with longer distance

skillsfactory, 2024-01-12:
I'll try
Thank you

closed 2024-03-03 by htmltiger