junkfix/esp32-ds18b20 #12: Reference to documentation

sirpete83, 2024-06-03:
Hi,
Maybe it is important, maybe not but the documentation of rmt module says:

Due to a software limitation in the GPIO driver, when both TX and RX channels are bound to the same GPIO, ensure the RX Channel is initialized before the TX Channel. If the TX Channel was set up first, then during the RX Channel setup, the previous RMT TX Channel signal will be overridden by the GPIO control signal.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/rmt.html

You did it the other way any it works. Eventually it is worth to look over this.

Best regards Peter

junkfix, 2024-06-03:
This is based on esp idf 4.x RMT driver which now deprecated. it does not have that information.
https://docs.espressif.com/projects/esp-idf/en/v4.4.7/esp32/api-reference/peripherals/rmt.html

idf 5.x is a new driver which i may port it later.

sirpete83, 2024-06-03:
Ah. Ok. Understand..
But maybe it is silicon related restriction.

junkfix, 2024-06-05:
This is rewritten with new api in the latest update