junkfix/esp32-ds18b20 #15: Dont compile for esp32c3

matsekberg, 2024-07-01:
`/Users/meg/Documents/Arduino/libraries/esp32-ds18b20/OneWireESP32.cpp: In constructor 'OneWire32::OneWire32(uint8_t, uint8_t, uint8_t, uint8_t)':
/Users/meg/Documents/Arduino/libraries/esp32-ds18b20/OneWireESP32.cpp:65:40: error: no match for 'operator=' (operand types are 'volatile gpio_dev_s::<unnamed union>' and 'int')
        GPIO.enable_w1ts = (0x1 << owpin);
                                        ^
In file included from /Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/soc/include/soc/gpio_periph.h:18,
                 from /Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/hal/include/hal/gpio_types.h:17,
                 from /Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/esp_hw_support/include/esp_sleep.h:13,
                 from /Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/cores/esp32/esp32-hal.h:33,
                 from /Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/cores/esp32/Arduino.h:36,
                 from /Users/meg/Documents/Arduino/libraries/esp32-ds18b20/OneWireESP32.cpp:15:
/Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/gpio_struct.h:61:11: note: candidate: 'gpio_dev_s::<unnamed union>& gpio_dev_s::<unnamed union>::operator=(const gpio_dev_s::<unnamed union>&)'
     union {
           ^
/Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/gpio_struct.h:61:11: note:   no known conversion for argument 1 from 'int' to 'const gpio_dev_s::<unnamed union>&'
/Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/gpio_struct.h:61:11: note: candidate: 'gpio_dev_s::<unnamed union>& gpio_dev_s::<unnamed union>::operator=(gpio_dev_s::<unnamed union>&&)'
/Users/meg/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/gpio_struct.h:61:11: note:   no known conversion for argument 1 from 'int' to 'gpio_dev_s::<unnamed union>&&'
/Users/meg/Documents/Arduino/libraries/esp32-ds18b20/OneWireESP32.cpp:67:13: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
        GPIO.enable1_w1ts.data = (0x1 << (owpin - 32));
             ^~~~~~~~~~~~
             enable_w1ts`

junkfix, 2024-07-01:
This looks like it is for this library version 1.0.7, upgrade the library to 2.0.0 and ALSO upgrade esp32 board from 2.0.17 to 3.x

I dont have C3 to test but the latest version of both above should work

matsekberg, 2024-07-02:
Yes I’m running 2.0.17 because of bugs in 3.x

matsekberg, 2024-07-02:
By the way I think I tested with the latest versions first as you suggested and that did not work.

junkfix, 2024-07-02:
> Yes I’m running 2.0.17 because of bugs in 3.x

try and edit line 65 from `GPIO.enable_w1ts = (0x1 << owpin);` to
```
GPIO.enable_w1ts.val = (0x1 << owpin);
```

closed 2024-07-04 by junkfix