junkfix/esp32-ds18b20 #17: Stack canary watchpoint triggered (tempTask)

GeorgeIoak, 2024-08-01:
I have a ESP32-S3 board and I'm using Arduino IDE 2.3.2 but right now I have core 2.0.17 installed.

I installed your 1.0.7 version of the code and tried to run the example and when I do I see:

```
ELF file SHA256: a6a25f9707fe092d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377410
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
entry 0x403c98d0
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (tempTask) 
Core  0 register dump:
PC      : 0x4037d963  PS      : 0x00060036  A0      : 0x8037ba8b  A1      : 0x3fcec590  
A2      : 0x3fcf07dc  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023  
A6      : 0x00060020  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00060023  A11     : 0x00000000  A12     : 0x00060020  A13     : 0x0000000c  
A14     : 0x02cf07dc  A15     : 0x00ffffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4037d960:0x3fcec590 0x4037ba88:0x3fcec5d0 0x4037bb40:0x3fcec610 0x40378be1:0x3fcec630 0x40378cb9:0x3fcec660 0x40378d88:0x3fcec680 0x4201832e:0x3fcec6b0 0x4201b355:0x3fcec9c0 0x42023295:0x3fcec9f0 0x4200e3ce:0x3fceca20 0x4200e463:0x3fceca50 0x42004d59:0x3fcecad0 0x42004e4e:0x3fcecb00 0x4200195a:0x3fcecb70 0x420017a1:0x3fcecbf0

ELF file SHA256: a6a25f9707fe092d
```

And if I decode the trace I get:

```
0x4037d960: compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
0x4037d960: spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
0x4037d960: xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:301
0x4037ba88: vPortEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
0x4037ba88: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1563
0x4037bb40: xQueueTakeMutexRecursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:731
0x40378be1: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:153
0x40378cb9: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:167
0x40378d88: __retarget_lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:323
```

GeorgeIoak, 2024-08-01:
To experiment I commented out the section in the task to read the temps.

I'm using `GPIO4`:
`OneWire32 ds(4, 0, 1, 0); //gpio pin, tx, rx, parasite power`

```E (2123) rmt: rmt_set_gpio(534): RMT CHANNEL ERR
E (2124) rmt: rmt_config(686): set gpio for RMT driver failed
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4200179c: a5201110 f01d0016 101cc1a2```

mathieucarbou, 2024-08-01:
Stack canary watchpoint triggered means that the stack size set to your task temp task is too smal

mathieucarbou, 2024-08-01:
Try to make use of uxTaskGetStackHighWaterMark to monitor the stack size of your tasks

GeorgeIoak, 2024-08-01:
OK, thanks, that's a new command I haven't see yet, thanks for pointing it out.

I'm just running your default example for now and trying to get it to work.

I did adjust the stack size all the way up to 10K but it would still crash.

What doesn't make sense to me is that the crashes happen when I comment out the temp readings:

```// 	for(;;){
// 		ds.request();
// 		vTaskDelay(750 / portTICK_PERIOD_MS);
// 		for(byte i = 0; i < MaxDevs; i++){
// 			uint8_t err = ds.getTemp(addr[i], currTemp[i]);
// 			if(err){
// 				const char *errt[] = {"", "CRC", "BAD","DC","DRV"};
// 				Serial.print(i); Serial.print(": "); Serial.println(errt[err]);
// 			}else{
// 				Serial.print(i); Serial.print(": "); Serial.println(currTemp[i]);
// 			}
// 		}
// 		vTaskDelay(3000 / portTICK_PERIOD_MS);
// 	}```

The code crashes but if I run the full example code it doesn't crash but it still has the RMT configuration error

```E (2124) rmt: rmt_set_gpio(534): RMT CHANNEL ERR
E (2125) rmt: rmt_config(686): set gpio for RMT driver failed```

so something still isn't happy.

Your using channels 0/1 so I don't see how this could be a problem:

`OneWire32 ds(4, 0, 1, 0); //gpio pin, tx, rx, parasite power`

mathieucarbou, 2024-08-01:
Note that I am not the author of the library, @junkfix is ;-) I happen to use it :-)