junkfix/esp32-ds18b20 #20: DS18B20 readings 3~4°C higher than actual :(

nadir-dogan, 2025-02-16:
Hi junkfix...
I have both original Dallas and Chinese DS18B20s.
I used them on Leonardo and on ESP32-C6 unfortunately everytime results is 3~4 C higher than actual temperature.
I'm working on this issue about one week. Not found any solution yet.
If do you have helpful knowledge I will be more happy.
My subject is not to do any calibration. I'm looking for correct native readings.
Thanks... #DS18B20

andrea1388, 2025-03-06:
Probably they are fake sensors.

nadir-dogan, 2025-03-07:
I see looks like the sensors self heated inside when always power on. What do you thing @andrea1388 about this situation?

HKPhysicist, 2025-04-29:
> I see looks like the sensors self heated inside when always power on. What do you thing [@andrea1388](https://github.com/andrea1388) about this situation?

Check your sensor with this library
https://github.com/koendv/CheckDS18B20/

andrea1388, 2025-04-29:
Try to measure the current absorption and check it with the data sheet

closed 2025-08-28 by junkfix